stack-of-tasks/tsid #210: undefined symbol error in libtsid_pywrap.so

Ruthrash, 2023-06-28:
Hi I am trying to use tsid for robots simulated in Isaacsim. I installed tsid using pip install with the [python interpreter](https://docs.omniverse.nvidia.com/app_isaacsim/app_isaacsim/tutorial_core_hello_world.html#:~:text=a%20Python%20interpreter%20for%20scripting.) provided by Isaacsim folks. 

I am facing an error similar to https://github.com/stack-of-tasks/pinocchio/issues/1229 as- 
```
>>> import tsid
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/home/ruthrash/.local/share/ov/pkg/isaac_sim-2022.2.1/kit/python/lib/python3.7/site-packages/cmeel.prefix/lib/python3.7/site-packages/tsid/__init__.py", line 1, in <module>
    from .libtsid_pywrap import *
ImportError: /home/ruthrash/.local/share/ov/pkg/isaac_sim-2022.2.1/kit/python/lib/python3.7/site-packages/cmeel.prefix/lib/python3.7/site-packages/tsid/libtsid_pywrap.so: undefined symbol: _ZN7eigenpy9NumpyType7getTypeEv

```
below are my outputs for:- 

1. ldd /home/ruthrash/.local/share/ov/pkg/isaac_sim-2022.2.1/kit/python/lib/python3.7/site-packages/cmeel.prefix/lib/python3.7/site-packages/tsid/libtsid_pywrap.so
```
	linux-vdso.so.1 (0x00007ffdce3d8000)
	libtsid.so => /home/ruthrash/.local/share/ov/pkg/isaac_sim-2022.2.1/kit/python/lib/python3.7/site-packages/cmeel.prefix/lib/python3.7/site-packages/tsid/../../../libtsid.so (0x00007f9fadef0000)
	libeigenpy.so => /home/ruthrash/.local/share/ov/pkg/isaac_sim-2022.2.1/kit/python/lib/python3.7/site-packages/cmeel.prefix/lib/python3.7/site-packages/tsid/../../../libeigenpy.so (0x00007f9fad68b000)
	libpinocchio.so => /home/ruthrash/.local/share/ov/pkg/isaac_sim-2022.2.1/kit/python/lib/python3.7/site-packages/cmeel.prefix/lib/python3.7/site-packages/tsid/../../../libpinocchio.so (0x00007f9fad654000)
	libboost_filesystem.so.1.81.0 => /home/ruthrash/.local/share/ov/pkg/isaac_sim-2022.2.1/kit/python/lib/python3.7/site-packages/cmeel.prefix/lib/python3.7/site-packages/tsid/../../../libboost_filesystem.so.1.81.0 (0x00007f9fad620000)
	libboost_system.so.1.81.0 => /home/ruthrash/.local/share/ov/pkg/isaac_sim-2022.2.1/kit/python/lib/python3.7/site-packages/cmeel.prefix/lib/python3.7/site-packages/tsid/../../../libboost_system.so.1.81.0 (0x00007f9fad61b000)
	liburdfdom_sensor.so.3.0 => /home/ruthrash/.local/share/ov/pkg/isaac_sim-2022.2.1/kit/python/lib/python3.7/site-packages/cmeel.prefix/lib/python3.7/site-packages/tsid/../../../liburdfdom_sensor.so.3.0 (0x00007f9fad612000)
	liburdfdom_model.so.3.0 => /home/ruthrash/.local/share/ov/pkg/isaac_sim-2022.2.1/kit/python/lib/python3.7/site-packages/cmeel.prefix/lib/python3.7/site-packages/tsid/../../../liburdfdom_model.so.3.0 (0x00007f9fad5df000)
	liburdfdom_world.so.3.0 => /home/ruthrash/.local/share/ov/pkg/isaac_sim-2022.2.1/kit/python/lib/python3.7/site-packages/cmeel.prefix/lib/python3.7/site-packages/tsid/../../../liburdfdom_world.so.3.0 (0x00007f9fad5aa000)
	libconsole_bridge.so.1.0 => /home/ruthrash/.local/share/ov/pkg/isaac_sim-2022.2.1/kit/python/lib/python3.7/site-packages/cmeel.prefix/lib/python3.7/site-packages/tsid/../../../libconsole_bridge.so.1.0 (0x00007f9fad5a3000)
	libtinyxml.so => /home/ruthrash/.local/share/ov/pkg/isaac_sim-2022.2.1/kit/python/lib/python3.7/site-packages/cmeel.prefix/lib/python3.7/site-packages/tsid/../../../libtinyxml.so (0x00007f9fad57c000)
	libhpp-fcl.so => /home/ruthrash/.local/share/ov/pkg/isaac_sim-2022.2.1/kit/python/lib/python3.7/site-packages/cmeel.prefix/lib/python3.7/site-packages/tsid/../../../libhpp-fcl.so (0x00007f9fad262000)
	libboost_serialization.so.1.81.0 => /home/ruthrash/.local/share/ov/pkg/isaac_sim-2022.2.1/kit/python/lib/python3.7/site-packages/cmeel.prefix/lib/python3.7/site-packages/tsid/../../../libboost_serialization.so.1.81.0 (0x00007f9fad21f000)
	libboost_chrono.so.1.81.0 => /home/ruthrash/.local/share/ov/pkg/isaac_sim-2022.2.1/kit/python/lib/python3.7/site-packages/cmeel.prefix/lib/python3.7/site-packages/tsid/../../../libboost_chrono.so.1.81.0 (0x00007f9fad213000)
	liboctomap.so => /home/ruthrash/.local/share/ov/pkg/isaac_sim-2022.2.1/kit/python/lib/python3.7/site-packages/cmeel.prefix/lib/python3.7/site-packages/tsid/../../../liboctomap.so (0x00007f9fad1a1000)
	liboctomath.so => /home/ruthrash/.local/share/ov/pkg/isaac_sim-2022.2.1/kit/python/lib/python3.7/site-packages/cmeel.prefix/lib/python3.7/site-packages/tsid/../../../liboctomath.so (0x00007f9fad199000)
	libeiquadprog.so => /home/ruthrash/.local/share/ov/pkg/isaac_sim-2022.2.1/kit/python/lib/python3.7/site-packages/cmeel.prefix/lib/python3.7/site-packages/tsid/../../../libeiquadprog.so (0x00007f9fad176000)
	libboost_python37.so.1.81.0 => /home/ruthrash/.local/share/ov/pkg/isaac_sim-2022.2.1/kit/python/lib/python3.7/site-packages/cmeel.prefix/lib/python3.7/site-packages/tsid/../../../libboost_python37.so.1.81.0 (0x00007f9fad12b000)
	libstdc++.so.6 => /lib/x86_64-linux-gnu/libstdc++.so.6 (0x00007f9facf32000)
	libm.so.6 => /lib/x86_64-linux-gnu/libm.so.6 (0x00007f9facde3000)
	libgcc_s.so.1 => /lib/x86_64-linux-gnu/libgcc_s.so.1 (0x00007f9facdc8000)
	libc.so.6 => /lib/x86_64-linux-gnu/libc.so.6 (0x00007f9facbd6000)
	libboost_atomic.so.1.81.0 => /home/ruthrash/.local/share/ov/pkg/isaac_sim-2022.2.1/kit/python/lib/python3.7/site-packages/cmeel.prefix/lib/python3.7/site-packages/tsid/../../../libboost_atomic.so.1.81.0 (0x00007f9facbcb000)
	librt.so.1 => /lib/x86_64-linux-gnu/librt.so.1 (0x00007f9facbbf000)
	libpthread.so.0 => /lib/x86_64-linux-gnu/libpthread.so.0 (0x00007f9facb9c000)
	libassimp.so.5 => /home/ruthrash/.local/share/ov/pkg/isaac_sim-2022.2.1/kit/python/lib/python3.7/site-packages/cmeel.prefix/lib/python3.7/site-packages/tsid/../../../libassimp.so.5 (0x00007f9fac172000)
	libdl.so.2 => /lib/x86_64-linux-gnu/libdl.so.2 (0x00007f9fac16c000)
	libutil.so.1 => /lib/x86_64-linux-gnu/libutil.so.1 (0x00007f9fac165000)
	/lib64/ld-linux-x86-64.so.2 (0x00007f9fae208000)
	libz.so.1 => /lib/x86_64-linux-gnu/libz.so.1 (0x00007f9fac149000)
```
2. ldd /home/ruthrash/.local/share/ov/pkg/isaac_sim-2022.2.1/kit/python/lib/python3.7/site-packages/cmeel.prefix/lib/python3.7/site-packages/tsid/../../../libeigenpy.so
```
	linux-vdso.so.1 (0x00007ffe311c1000)
	libboost_python37.so.1.81.0 => /home/ruthrash/.local/share/ov/pkg/isaac_sim-2022.2.1/kit/python/lib/python3.7/site-packages/cmeel.prefix/lib/libboost_python37.so.1.81.0 (0x00007f5b5c13d000)
	libstdc++.so.6 => /lib/x86_64-linux-gnu/libstdc++.so.6 (0x00007f5b5bf46000)
	libm.so.6 => /lib/x86_64-linux-gnu/libm.so.6 (0x00007f5b5bdf7000)
	libgcc_s.so.1 => /lib/x86_64-linux-gnu/libgcc_s.so.1 (0x00007f5b5bdda000)
	libc.so.6 => /lib/x86_64-linux-gnu/libc.so.6 (0x00007f5b5bbe8000)
	librt.so.1 => /lib/x86_64-linux-gnu/librt.so.1 (0x00007f5b5bbde000)
	libdl.so.2 => /lib/x86_64-linux-gnu/libdl.so.2 (0x00007f5b5bbd8000)
	libpthread.so.0 => /lib/x86_64-linux-gnu/libpthread.so.0 (0x00007f5b5bbb5000)
	libutil.so.1 => /lib/x86_64-linux-gnu/libutil.so.1 (0x00007f5b5bbb0000)
	/lib64/ld-linux-x86-64.so.2 (0x00007f5b5c9ef000)
```

Please help me solve this issue. Many thanks

nim65s, 2023-06-28:
Hi @Ruthrash,

Thanks for this report. This issue comes from the fact that we are dropping support for python 3.7, and I stopped running tests here (upstream support actually ended [yesterday](https://peps.python.org/pep-0537/)).

I will try to provide a last working version (following https://github.com/stack-of-tasks/pinocchio/issues/1937) , but you should seriously consider upgrading your dependencies.

Meanwhile, the workaround provided in pinocchio will also work here, ie. `python -m pip install 'eigenpy<3'`

Ruthrash, 2023-06-30:
Thanks a lot for your answer, @nim65s! This solved my issue

closed 2023-06-30 by Ruthrash