stack-of-tasks/tsid #214: Large time steps

erik-gattung, 2023-10-13:
Hello team,

I am using the tsid Python API for running an example. By default, the time step dt is set to 0.005. However, I want to increase the value of dt. But even for values like dt = 0.02, the optimization fails (Error code: 1). Why is that?

andreadelprete, 2023-10-13:
Hi @erik-gattung , 

can you tell me which test is failing? Typically the optimization fails if the constraints are unfeasible, so you can try removing some constraints to see if things improve.

erik-gattung, 2023-10-14:
Hi @andreadelprete,

unfortunately, the program doesn't tell me what caused the failure, only that the third iteration of the optimization is failing.
Also the example I'm running is rather minimal. It features only an end-effector and a postural task and enforcing torque and pos/vel limits. Removing the posture task also leads to a failure. Removing the other tasks makes the example dysfunctional.
Is it possible that the optimization is just not designed for low frequencies? Currently, the new joint positions are calculated by integrating the acceleration and velocity after every iteration. Given a large enough step size, can this lead to joint values which are inadmissible?

andreadelprete, 2023-10-16:
If I understood correctly you have 3 constraints:
- joint torques
- joint positions
- joint velocities

Joint velocities are rarely an issue. My guess is that the problem stems from the incompatibility of torque and position limits. Can you give me the details of how you are inserting the joint position limits in the problem? Are you using [this class](https://github.com/stack-of-tasks/tsid/blob/master/include/tsid/tasks/task-joint-posVelAcc-bounds.hpp)? If so, that's already the best you can do, but that's not guaranteed to lead to feasible problems. The method we implemented for handling joint position bounds [1] ensures consistency with the joint acceleration bounds, but not with the joint torque bounds. If you can replace the joint torque bounds with joint acceleration bounds then you should be fine. Otherwise, things are trickier, and I can suggest you to read [2] to better understand the problem. 

[1] Andrea del Prete. Joint Position and Velocity Bounds in Discrete-Time Acceleration/Torque Control of Robot Manipulators. IEEE Robotics and Automation Letters, IEEE 2018, 3 (1), pp.281-288.￿10.1109/LRA.2017.2738321￿. hal-01356989v3
[2] VBOC: Learning the Viability Boundary of a Robot Manipulator using Optimal Control, https://arxiv.org/abs/2305.07535

erik-gattung, 2023-10-29:
Hi,

disregarding the joint torques does help finding an admissible solution. However, a few iterations later, something else causes the optimization to find no solution. So, I am just considering using a workaround and staying with the relatively small time step size.
Thanks anyway!

closed 2023-10-29 by erik-gattung