stack-of-tasks/tsid #223: Does it contain c++ examples?

Mr-Red331, 2024-03-06:
Hi,

Thank you for providing such a great repository!

I noticed that all the examples and tests are based on Python, does this project have any examples in C++ for us to learn from that I haven't noticed?

andreadelprete, 2024-03-06:
Yes, I think this one is the best starting point: https://github.com/stack-of-tasks/tsid/blob/master/tests/tsid-formulation.cpp

Mr-Red331, 2024-03-06:
> Yes, I think this one is the best starting point: https://github.com/stack-of-tasks/tsid/blob/master/tests/tsid-formulation.cpp

Thanks for solving my confusion.

I've never used the Boost library before, so I'm a bit confused. In the `tsid-formulation.cpp` you mentioned, it provides a class named `StandardRomeoInvDynCtrl` and provides 4 `BOOST_AUTO_TEST_CASE` test examples. These 4 examples are 4 demonstrations that can be run separately in the `main`, am I understanding this correctly?

andreadelprete, 2024-03-06:
You can ignore anything boost related in that file. That's just because this file contains unit tests, but boost is not necessary for using tsid.

Mr-Red331, 2024-03-07:
I understand! Thank you for your detailed answer.

closed 2024-03-07 by Mr-Red331