stack-of-tasks/tsid #232: build issue with catkin build ROS1

rua0ra1, 2024-07-10:
Hi,
I am trying to build tsid repo with  catkin build ROS1. But, I  am having problem with eigenpy as shown in figure
[image]
But I have already installed the eigenpy  by using
sudo apt-get install ros-noetic-eigenpy

I really appreciate if you can help me to solve this issue.

rua0ra1, 2024-07-11:
sorry @nim65s it didn't work. Are there any build instructions for ros?. Ideally I want to use tsid c++ library in ros to build a hierarchical control for our robot. But, So far, I haven't seen any build instructions for ros . And also I have seen examples in python. Are there any examples for rosc++ or c++?. Am I missing something here?. Thank you and I really appreciate your help

nim65s, 2024-07-13:
Standard CMake build instructions seems to work fine for me on ros noetic. At least this work without issue:
```Dockerfile
FROM ros:noetic

RUN --mount=type=cache,sharing=locked,target=/var/cache/apt \
    --mount=type=cache,sharing=locked,target=/var/lib/apt \
    apt-get update -y && DEBIAN_FRONTEND=noninteractive apt-get install -qqy --no-install-recommends \
    ros-noetic-pinocchio

ADD https://github.com/stack-of-tasks/tsid/releases/download/v1.7.0/tsid-1.7.0.tar.gz /
ADD https://github.com/stack-of-tasks/eiquadprog/releases/download/v1.2.8/eiquadprog-1.2.8.tar.gz /
WORKDIR /src

RUN . /opt/ros/noetic/setup.sh \
 && tar xvf ../eiquadprog-1.2.8.tar.gz \
 && cmake -B build -S eiquadprog-1.2.8 \
 && cmake --build build \
 && cmake --build build -t test \
 && cmake --build build -t install \
 && rm -rf build eiquadprog-1.2.8 \
 && tar xvf ../tsid-1.7.0.tar.gz \
 && cmake -B build -S tsid-1.7.0 \
 && cmake --build build \
 && cmake --build build -t test \
 && cmake --build build -t install \
 && rm -rf build tsid-1.7.0

rua0ra1, 2024-07-13:
Hi @nim65s , thanks for the help. I was trying to build and use tsid as a catkin package.

```sh
# Step 1: Create a Catkin workspace
cd ~
mkdir -p catkin_ws/src

# Step 2: Clone the TSID repository into the src directory
cd ~/catkin_ws/src
git clone https://github.com/stack-of-tasks/tsid.git

# Step 3: Build the TSID package
cd ~/catkin_ws
catkin build tsid

I installed tsid using robotpkg-py3*-tsid earlier. I went through the python examples without having any issue.

Do you think previous installation using robotpkg conflicting with building with catkin package. Thank you

nim65s, 2024-07-15:
You must not have the same package installed twice. You can either use our softwares from robotpkg- or from ros-, but not at the same time. Having 2 different versions of eigenpy / hpp-fcl / pinocchio / eiquadprog / tsid / etc. active at the same time will result in errors.

For your setup, I don't have catkin on ros noetic, but catkin_make / catkin_make_isolated. And these are working fine:
```Dockerfile
FROM ros:noetic

RUN --mount=type=cache,sharing=locked,target=/var/cache/apt \
    --mount=type=cache,sharing=locked,target=/var/lib/apt \
    --mount=type=cache,sharing=locked,target=/root/.cache \
    apt-get update -y && DEBIAN_FRONTEND=noninteractive apt-get install -qqy --no-install-recommends \
    git \
    ros-noetic-eiquadprog \
    ros-noetic-pinocchio

WORKDIR /ws/src
RUN . /opt/ros/noetic/setup.sh \
 && git clone --recursive https://github.com/stack-of-tasks/tsid \
 && cd .. \
 && catkin_make_isolated

rua0ra1, 2024-07-15:
@nim65s thank you so much for detail answer and I really appreciate it. I will follow the setup instructions given above. Sorry for the trouble.

closed 2025-03-17 by nim65s