stack-of-tasks/tsid #243: Installation through Debian packages

andreadelprete, 2024-10-28:
Hi, 
a student of mine is having troubles installing TSID through Debian packages. He's working on WLS, with Ubuntu 22.04. He followed the instructions on Pinocchio's website for configuring robotpkg, but when installing TSID he gets the following error:
```
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 robotpkg-py310-tsid : Depends: robotpkg-tsid (= 1.6.1) but 1.7.1 is to be installed or
                                robotpkg-tsid (= 1.6.2) but 1.7.1 is to be installed or
                                robotpkg-tsid (= 1.6.3) but 1.7.1 is to be installed or
                                robotpkg-tsid (= 1.7.0) but 1.7.1 is to be installed
                       Depends: robotpkg-py310-pinocchio (= 2.6.10) but 3.2.0 is to be installed or
                                robotpkg-py310-pinocchio (= 2.6.14) but 3.2.0 is to be installed or
                                robotpkg-py310-pinocchio (= 2.6.21) but 3.2.0 is to be installed or
                                robotpkg-py310-pinocchio (= 2.6.17) but 3.2.0 is to be installed or
                                robotpkg-py310-pinocchio (= 3.0.0) but 3.2.0 is to be installed or
                                robotpkg-py310-pinocchio (= 2.6.7) but 3.2.0 is to be installed or
                                robotpkg-py310-pinocchio (= 2.6.20) but 3.2.0 is to be installed or
                                robotpkg-py310-pinocchio (= 2.7.0) but 3.2.0 is to be installed or
                                robotpkg-py310-pinocchio (= 2.6.9) but 3.2.0 is to be installed or
                                robotpkg-py310-pinocchio (= 2.6.5) but 3.2.0 is to be installed or
                                robotpkg-py310-pinocchio (= 2.6.15) but 3.2.0 is to be installed or
                                robotpkg-py310-pinocchio (= 3.0.0r1) but 3.2.0 is to be installed or
                                robotpkg-py310-pinocchio (= 2.6.8r1) but 3.2.0 is to be installed or
                                robotpkg-py310-pinocchio (= 2.6.8) but 3.2.0 is to be installed
                       Depends: robotpkg-py310-eigenpy (= 3.1.1) but 3.10.0 is to be installed or
                                robotpkg-py310-eigenpy (= 2.7.10) but 3.10.0 is to be installed or
                                robotpkg-py310-eigenpy (= 3.3.0) but 3.10.0 is to be installed or
                                robotpkg-py310-eigenpy (= 3.1.4) but 3.10.0 is to be installed or
                                robotpkg-py310-eigenpy (= 3.5.0) but 3.10.0 is to be installed or
                                robotpkg-py310-eigenpy (= 2.7.13) but 3.10.0 is to be installed or
                                robotpkg-py310-eigenpy (= 2.8.0) but 3.10.0 is to be installed or
                                robotpkg-py310-eigenpy (= 3.4.0) but 3.10.0 is to be installed or
                                robotpkg-py310-eigenpy (= 2.9.1) but 3.10.0 is to be installed or
                                robotpkg-py310-eigenpy (= 2.9.0) but 3.10.0 is to be installed or
                                robotpkg-py310-eigenpy (= 2.7.11) but 3.10.0 is to be installed or
                                robotpkg-py310-eigenpy (= 2.9.2) but 3.10.0 is to be installed or
                                robotpkg-py310-eigenpy (= 2.7.12) but 3.10.0 is to be installed or
                                robotpkg-py310-eigenpy (= 3.7.0) but 3.10.0 is to be installed
                       Depends: robotpkg-pinocchio (= 2.6.17) but 3.2.0 is to be installed or
                                robotpkg-pinocchio (= 2.6.14) but 3.2.0 is to be installed or
                                robotpkg-pinocchio (= 2.7.0) but 3.2.0 is to be installed or
                                robotpkg-pinocchio (= 2.6.8) but 3.2.0 is to be installed or
                                robotpkg-pinocchio (= 2.6.5) but 3.2.0 is to be installed or
                                robotpkg-pinocchio (= 3.0.0r1) but 3.2.0 is to be installed or
                                robotpkg-pinocchio (= 2.6.21) but 3.2.0 is to be installed or
                                robotpkg-pinocchio (= 2.6.8r1) but 3.2.0 is to be installed or
                                robotpkg-pinocchio (= 3.0.0) but 3.2.0 is to be installed or
                                robotpkg-pinocchio (= 2.6.7) but 3.2.0 is to be installed or
                                robotpkg-pinocchio (= 2.6.9) but 3.2.0 is to be installed or
                                robotpkg-pinocchio (= 2.6.10) but 3.2.0 is to be installed or
                                robotpkg-pinocchio (= 2.6.15) but 3.2.0 is to be installed or
                                robotpkg-pinocchio (= 2.6.20) but 3.2.0 is to be installed
                       Depends: robotpkg-hpp-fcl+doc (= 1.8.1) but it is not going to be installed or
                                robotpkg-hpp-fcl+doc (= 2.3.0) but it is not going to be installed or
                                robotpkg-hpp-fcl (= 2.4.0) or
                                robotpkg-hpp-fcl (= 2.3.6) or
                                robotpkg-hpp-fcl (= 2.2.0) or
                                robotpkg-hpp-fcl+doc (= 2.2.0) but it is not going to be installed or
                                robotpkg-hpp-fcl+doc (= 2.4.0) but it is not going to be installed or
                                robotpkg-hpp-fcl+doc (= 2.3.5) but it is not going to be installed or
                                robotpkg-hpp-fcl+doc (= 2.4.4) but it is not going to be installed or
                                robotpkg-hpp-fcl+doc (= 1.8.0) but it is not going to be installed or
                                robotpkg-hpp-fcl+doc (= 2.3.6) but it is not going to be installed or
                                robotpkg-hpp-fcl (= 2.3.5) or
                                robotpkg-hpp-fcl+doc (= 2.4.1) but it is not going to be installed or
                                robotpkg-hpp-fcl (= 2.3.0) or
                                robotpkg-hpp-fcl (= 2.4.1) or
                                robotpkg-hpp-fcl (= 1.8.1) or
                                robotpkg-hpp-fcl (= 2.4.4) or
                                robotpkg-hpp-fcl (= 1.8.0)
                       Depends: robotpkg-eiquadprog (= 1.2.5) but 1.2.9 is to be installed or
                                robotpkg-eiquadprog (= 1.2.3) but 1.2.9 is to be installed or
                                robotpkg-eiquadprog (= 1.2.8) but 1.2.9 is to be installed or
                                robotpkg-eiquadprog (= 1.2.4) but 1.2.9 is to be installed or
                                robotpkg-eiquadprog (= 1.2.6) but 1.2.9 is to be installed
E: Unable to correct problems, you have held broken packages.
```

Any idea what could be wrong?

nim65s, 2024-10-28:
Hi Andrea,

Sorry, this is my fault, I updated the package recently, and did not notice the mail with that build failure:
http://robotpkg.openrobots.org/rbulk/robotpkg/optimization/py-tsid/py310-tsid-1.7.1/Ubuntu-22.04-x86_64/bulk.html

It is a dump `__init__.py` file missing, I'll fix it.

andreadelprete, 2024-10-29:
Thanks @nim65s . Can you let me know when this will be fixed on robotpkg so that the student can re-try the installation?

nim65s, 2024-10-30:
Hi @andreadelprete
This is fixed: http://robotpkg.openrobots.org/rbulk/robotpkg/optimization/py-tsid/

PR #244 (2024-10-30): CMake: fix python setup
fix #243

I broke that in latest release, sorry.

This should have been caught by the Nix CI which try to import the python package after installation, but actually since [PEP 420](https://peps.python.org/pep-0420/) that `tsid` folder without a `__init__.py` file was considered a valid namespace package, so no error were raised.

It also  should have been caught by my robotpkg tests during the release process, but this did only raise a warning in the test setup, while raising a fatal error at the DEB package generation step, which is not part of the tests.
commit c9a118b4 (2024-10-30): CMake: fix python setup

fix #243
closed 2024-10-30 by nim65s