stack-of-tasks/tsid #247: Expose TaskContactForceEquality python binding

danielcostanzi18, 2024-12-20:
Hello everyone,

First off, thanks a lot for the great work on TSID!
I want to use TSID to control the force at the end effector of a manipulator, to do so I want to use the task `TaskContactForceEquality`. The problem is that it's python binding is not exposed.
I tried to naively expose it with [these](https://github.com/danielcostanzi18/tsid/commit/b583cc2ea92355501debc78e0114aaaa6d4b2d6f) changes.

When I try to instantiate the task I get the following error:
> ArgumentError: Python argument types in
    TaskContactForceEquality.__init__(TaskContactForceEquality, str, RobotWrapper, float, Contact6d)
did not match C++ signature:
    __init__(_object*, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > name, tsid::robots::RobotWrapper {lvalue} robot, double dt, tsid::contacts::ContactBase {lvalue} contact)

It seems that the problem lies in the type of `contact` that I am passing, which is of type `tsid.tsid_pywrap.Contact6d`. From the C++ signature, it appears that the constructor expects `tsid::contacts::ContactBase` instead.

Could anyone shed light on how I might resolve this mismatch or if there's an alternative way to expose the correct binding for Contact6d?

If you want to reproduce the issue you can launch the example that I prepared [exercizes/ex_5_ur5_force_task.py](https://github.com/danielcostanzi18/tsid/commit/c27be4b857e11c12cebc15afb28f2d9cdba5d248).

Thanks in advance for your help!

Best regards,
Daniel

andreadelprete, 2025-01-02:
Hi @danielcostanzi18 
it seems to me the problem is that in Python `Contact6d` is not recognized as a `ContactBase`, even though it is. Indeed if you take a look at how other Python bindings have been implemented (such as [this one](https://github.com/danielcostanzi18/tsid/blob/master/include/tsid/bindings/python/formulations/formulation.hpp)) you need to implement a different method for each possible child class of the input argument. In your case, you would need a constructor for each possible child of `ContactBase` (such as `Contact6d`).

danielcostanzi18, 2025-01-07:
Hi @andreadelprete 
Thanks a lot for your help, I'm now able to instantiate the task by passing a `Contact6d`.
I'd like to open a PR, but first would probably be better to have a working example, what do you think?
For that matter, I'd open another issue.

andreadelprete, 2025-01-07:
Yes that would be ideal thanks!